CHERIoT-Platform/cheriot-rtos #56: Should interrupts be enabled when the scheduler yields?

rmn30, 2023-04-14:
As of the merge of #44 002271096c73ba8fa39334c70770ad911928c055 disabled interrupts when yielding in the scheduler. 
It's not clear whether the comment on `yield_interrupt_enabled` which describes the need for enabling interrupts still applies.
A simple test of a thread calling `thread_sleep` with interrupts disabled worked as expected: the thread was suspended, yielded and the idle thread was scheduled allowing timer interrupts to be received.

We should think about this and, at the very least, fix the comment.

rmn30, 2024-06-03:
I still don't understand the comment and it is now even more confusing as the function does not enable interrupts. If you think it's fine as is we should rename the function to just `yield` and remove the comment.

davidchisnall, 2024-06-03:
I think the comment is wrong and we should rename the function.