CHERIoT-Platform/cheriot-rtos #58: Document and enforce stack size limits

rmn30, 2023-04-14:
Stacks larger than about 8 KiB are problematic for compartment switches because the switcher cannot guarantee that the remainder stack has a representable length.
In particular this [`csetboundsexact`](https://github.com/microsoft/cheriot-rtos/blob/e24f37fafa05d8eb23f81c76ee3b997c6b9415b3/sdk/core/switcher/entry.S#L184) might trap.
Given that stack bounds must be 16-byte aligned and the capability precision is 9 bits stacks up to `16 * 511 = 8176` bytes should be fine.

We should document this limit and possibly enforce it in the xmake and / or linker.

davidchisnall, 2023-05-16:
Some of this code wants refactoring anyway.  I don't really like the fact that we're passing the thread init structures into the loader at compile time, because it means that we can't have a single loader binary and that makes auditing hard.  I'd like to move the initial thread state into declarative data so that it's checkable from the linker's report.

davidchisnall, 2023-07-10:
This refactoring has been done, so this should now be easy.

commit c6f094e1 (2023-07-12): Enforce stack-size limit.

The compartment switcher will not behave correctly if subsetting a stack
results in something unrepresentable.  We can avoid this by limiting
stacks to the maximum representable size for 16-byte alignment.

Eventually this can be removed if we make the switcher round the shared
bit of the stack down to avoid overlap.  At the moment, it's not a
serious limitation because our stacks are very rarely over 1 KiB and
the limit is about 8 KiB.

Fixes #58
PR #96 (2023-07-12): Enforce stack-size limit.
The compartment switcher will not behave correctly if subsetting a stack results in something unrepresentable.  We can avoid this by limiting stacks to the maximum representable size for 16-byte alignment.

Eventually this can be removed if we make the switcher round the shared bit of the stack down to avoid overlap.  At the moment, it's not a serious limitation because our stacks are very rarely over 1 KiB and the limit is about 8 KiB.

Fixes #58
commit 7038b601 (2023-07-12): Enforce stack-size limit. (#96)

The compartment switcher will not behave correctly if subsetting a stack
results in something unrepresentable. We can avoid this by limiting
stacks to the maximum representable size for 16-byte alignment.

Eventually this can be removed if we make the switcher round the shared
bit of the stack down to avoid overlap. At the moment, it's not a
serious limitation because our stacks are very rarely over 1 KiB and the
limit is about 8 KiB.

Fixes #58
closed 2023-07-12 by davidchisnall