CHERIoT-Platform/cheriot-rtos #140: Create sub-quotas for allocation capabilities

davidchisnall, 2023-11-29:
The allocator currently deals with static heap quotas.  Sometimes it's useful to allow another compartment to allocate memory on your behalf, but not to consume your entire quota.  We should allow a mechanism for creating a new quota object that contains part of an existing quota.  This should have the same ID as the original, so that objects can be freed in both.

The main complication in this is that, currently, quotas are checked without the allocator lock being held (I think).  We need to make sure that the mechanism is robust in the presence of a quota being deallocated during an allocation or deallocation operation.

hlef, 2024-10-10:
We are encountering an issue in the network stack which would likely benefit from such a feature.

- Sockets are allocated with the quota of the TCP/IP compartment. See [here](). This would be hard to change since this allocation and the corresponding free are done deep in the internals of FreeRTOS+TCP.
- To prevent the caller from DoS-ing the TCP/IP compartment by exhausting its quota through creating many sockets without freeing them, we claimed the allocation with the quota of the caller. See [here](). Thus the caller is accountable for the socket allocation, and would exhaust its own quota in the process.
- Unfortunately this approach does not work, since the caller can always call `heap_free_all` to remove this claim. The problem would also be there if the caller simply had a much larger quota than the TCP/IP stack.

To address this issue, it would be nice if we could have a mechanism to loan quota: the caller would give enough quota to allocate the socket to the TCP/IP stack, and the loan would be returned when the socket is freed.

This does raise questions on how to refund the quota though:

- The free of the socket happens asynchronously, deep in FreeRTOS+TCP. There is no callback to our code to say that the socket was freed.
- We could refund the quota when closing the socket, and claim it with the TCP/IP stack compartment's quota. If the TCP/IP stack's quota is insufficient we would simply not close the socket. We would thus allow callers to consume some of the stack's quota, but only for a short and rather well-defined amount of time.

tolbooth, 2025-11-28:
So, I've taken a look at this issue, and unfortunately it doesn't seem like there's a simple solution.

First off, it's not entirely clear that these quotas *should* share an ID. The issue with this model is that `heap_free_all` depends directly on said ID, meaning:

1. We cannot support untrusted delegatees. Compartment A may want to hand out quota to compartment B to make allocations, but not allow B to arbitrarily free A's own allocations. If we assume a fully trusted recipient, we should have no trouble handing them the original quota anyway.
2. Clean reset of compartments holding delegated sub-quotas gets complicated. To perform a compartment reset, we'd like to be able to invoke `heap_free_all` on the sub-quota to release all of its allocations. If the sub-quota and parent quota share an ID, this action would also affect the parent.

It seems to me, then, that it would be nice to be able to distinguish between parent and child quotas. This, however, raises an issue.

### The Issue

These quota IDs are stored as 16-bit integers. Allocations are associated with a quota by storing this ID in the `MChunkHeader` metadata. Among other things, this is how `heap_free_all` distinguishes between allocations it should and shouldn't free as it walks the heap.

In the `MChunkHeader`, 3 bits of this integer are used for other purposes, meaning we really only have **13 bits** to represent this unique ID. ID 0 is reserved for "no owner", capping us at 8191 possible uniquely identifiable quotas.

8K unique IDs for dynamically allocated quotas does not seem like enough. We can't re-use IDs between parent and child, as described above, since it would leave a hole in the trust model. Given the mechanism by which IDs are allocated (`++owner_id`), allowing programmatic allocation of quotas could very quickly burn through our small ID space. Even if the usage of such an API were heavily restricted, this could very easily put the system in an unrecoverable state.

It's also not all that easy to support ID re-use either. At minimum we'd have to keep around something like a bitmap to track free/in-use IDs. Given the memory budget, spending 1KB on such a structure for 8K IDs doesn't seem all too appealing.

### Reduce the ID Space

The main downside of supporting dynamic allocation is that ID reuse requires a structure at least as expensive (in the worst case) as a bitmap. Then a potential solution is to massively decrease the total pool of IDs. Given a pool of only, say, 128 unique IDs, we would need just 16 bytes of bitmap. The overhead of such a small bitmap would likely be an easier pill to swallow than the 1KB required for all 8K? Further, this approach allows for recovery from exhausted ID space, as quotas could be forcibly freed and their IDs recycled.

However, such a small ID space could be very limiting. Evaluating this option further requires answering questions like: How many quota-bearing compartments do we want to support? Are there few enough long running operations requiring sub-quotas that this wouldn't be an issue? Perhaps a topic for future discussion.

### No Dynamic Allocation

Instead of allowing for true 'sub-quotas', derived from a parent dynamically, we could instead support an API for moving quota between allocator capabilities. I believe @davidchisnall has mentioned something like this before.

In this model, quotas would remain statically allocated. Let's take Hugo's network stack example: compartments that expect to create a socket (or many), would declare a pool of quotas with 0 capacity. The compartment would first move some portion of its main quota to one in this pool, then delegate the quota. At any point, the parent could reclaim the space with a `heap_free_all`, and another call to move. The downside here being that it wouldn't provide the ability to 'lend' quota in the sense of automatic return. We'd also be limited in how many in-flight sub-quotas a given compartment can have.

This API gets tricky around compartment boundaries: if we allow movement of quota arbitrarily, i.e. from an allocator capability belonging to compartment A to an allocator capability belonging to compartment B, then a malicious recipient could permanently steal this quota. In case of a compartment crash, we'd be left in a potentially unrecoverable situation.

There are two options I can see here:

1. We can use the third, currently unused permission bit on the `AllocatorCapability` as `AllocatorPermitMoveFrom`. Permission to move *to* a quota would be implicit, and we enforce that delegated quotas have this permission stripped. This lets us sidestep the issue of cross-compartment moves being exploited by malicious recipients. This is also tied to a particular handle, and so allows for more flexibility than barring cross-compartment moves outright.
2. We implement a way of tying a quota to the compartment it originates from, in a way that's visible to the allocator at runtime. With that in place, `move` could be restricted to function only on quotas from the same compartment.

We can also use a combination of both approaches to express limited quota lending functionality. A compartment passes both the sub-quota and a fully depermissioned handle to its main quota. The receiving compartment can move quota from the sub-quota back into the overall usable pool of the parent any time it sees fit, but moves in the opposite direction are disallowed. The downsides of this approach are the static pool size, and the la [... cut ...] he address of the `CompartmentHeader`, since it's unique per compartment, and accessible at this point.
https://github.com/CHERIoT-Platform/cheriot-rtos/blob/3e59a54dfbc1d2d7a40ca7feebf121ca5a33c674/sdk/core/loader/boot.cc#L658-L665

Which could live in `reserved[0]`:
https://github.com/CHERIoT-Platform/cheriot-rtos/blob/3e59a54dfbc1d2d7a40ca7feebf121ca5a33c674/sdk/include/stdlib.h#L34-L42

I'd be interested in further opinions here. Neither approach (static, dynamic) is without its compromises.