CHERIoT-Platform/cheriot-rtos #204: Better protect against stack-overflow attacks

hlef, 2024-04-18:
Threads have a single stack that gets subdivided as they cross compartment boundaries.

This means that a compartment controls how much stack space another compartment has before calling it.

If this stack space is not checked at the compartment entry point, this creates an exploitable attack vector.

**Example**: if the total stack size if 4KB, a compartment A can maliciously consume exactly 3.96KB of stack to trigger a stack overflow at a very precise point in compartment B to corrupt its state and mount subsequent attacks.

This is a real problem for the TCB, e.g., the allocator or the scheduler.

Currently, we have a mechanism to thwart such attacks: the export table entry contains the stack required for the first function that's called. This number is easy to compute and makes sense as a bare minimum.

However, this number will remain vastly insufficient for most real-world cases, because it does not account for any other functions that first entry point may call.

Looking forward, we should:

1. Make it possible to specify arbitrary minimum stack requirements at an entry point granularity through a function attribute. Computing the worst case would be tricky, and potentially undesirable as it is effectively a policy that we do not want to hardcode in the compiler.

2. Go through TCB entry points (scheduler, allocator), and experimentally determine a safe value for this new attribute.

3. Discuss this issue in the book (and how to solve it with the attribute!)

davidchisnall, 2024-04-18:
We should also expose an API for querying the SHWM so that compartments can find out how much stack they've used across a call.

davidchisnall, 2024-04-29:
Fixed last week.

closed 2024-04-29 by davidchisnall