CHERIoT-Platform/cheriot-rtos #243: exercises/01 fails when run in docker against ibex

PhilDay-CT, 2024-06-11:
As noted at the Sonata Hackathon exercise/01.compartmentalisation doesn't run in the dev docker, although it does work if compiled and run on the Sonata board.   I tested against the latest commit (771f2bb - Jun 3rd) in case any of the recent changes to the dev contained had fixed it, but it still fails for me.

Digging in a little the line that fails in Ibex is generating the secret in secret.cc 
```
    secret = static_cast<int32_t>(rdcycle64());
```

Replacing this with, for example, makes everything run fine
```
    //secret = static_cast<int32_t>(rdcycle64());
    secret=666;	
```

Commenting out the code in secret.cc which waits for the uart (so the call to rdcycle64()  is in effect the first call in the therad) also makes no difference.

Adding a compartment_error_handler() gives the error as:
```
JavaScript compartment: Detected PermitAccessSystemRegistersViolation(0x18).  Register PCC(0x20) contained invalid value: 0x2004ba08 (v:0 0x2004a7c0-0x2004cb40 l:0x2380 o:0x0 p: G R-cgm- X- ---)
```

Calling rdcycle64() in any of the examples with IBEX works fine, so I'm assuming there is some difference in the build that I'm missing, but the only thing I spotted is that exercise xmake.luna has:
```
option("board")
    set_default("ibex-safe-simulator")
```
whereas the examples use
```
option("board")
    set_default("sail")
```

I'm happy to dig further  if someone can point me in the right direction.

davidchisnall, 2024-06-11:
I suspect that the recent(ish) changes to the AccessSystemRegisters permissions in Ibex may have broken `mcycleh` access.  The test suite uses `mcycle` but not `mcycleh` (it doesn't run long enough to overflow a 32-bit cycle counter.  @kliuMsft, can you check?

PhilDay-CT, 2024-06-11:
> If the board is `sail` then they're not using the Ibex simulator, they're the instruction simulator generated from the Sail formal model.

Ah, Ok thanks.  I tried changing one of the examples to the ibex simulator and then doing "xmake config --sdk=/cheriot-tools/; xmake clean; xmake; xmake run" - but that didn't fail which made me think this was the wrong track.  Looks like I need to "rm -rf  .xmake .cache" to really get back to a clean state ?

And the exercise needs the simulator because the sail model doesn't have a uart - is that right ?   

(Sorry for the questions - still feeling my way around here some)

davidchisnall, 2024-06-11:
You shouldn't need it, but sometimes `rm -rf .xmake build` is necessary.

davidchisnall, 2024-06-11:
> And the exercise needs the simulator because the sail model doesn't have a uart - is that right ?

The Sail UART is output only, yes.

davidchisnall, 2024-06-14:
Kunyan confirms that this is an Ibex bug.  The refactoring of the PermitAccessSystemRegisters logic accidentally missed `mcycleh`.

davidchisnall, 2024-06-15:
The dev container is updated with the fix so this should work now.

PhilDay-CT, 2024-06-16:
Yep - runs fine now, thanks

closed 2024-06-16 by PhilDay-CT