CHERIoT-Platform/cheriot-rtos #256: Tag violation when creating a std::string from string literal over 14 characters in length

PhilDay-CT, 2024-06-19:
Hit this when working on #255 

If in test_runner.cc lines  111-112
```
        const std::string S = "I am a walrus"s;
	debug_log("Trying to print std::string: {}", S);
```
are changed to 
```
        const std::string S = "I am the walrus"s;
	debug_log("Trying to print std::string: {}", S);
```

it results in 
```
Test runner: Trying to print std::string: Test runner: mcause: 0x1c, pcc: 0x80006e30 (v:0 0x80006d90-0x800075e8 l:0x858 o:0x0 p: G R-cgm- X- ---)
Test runner: Error TagViolation(0x2) in register CA1(0xb)
Test runner: Current test crashed
```

I tried to create a simple reproducer based on examples/01.hello_world, but it looks like I'm hitting another error in the error handler as
```
#include <compartment.h>
#include <debug.hh>
#include <string>

/// Expose debugging features unconditionally for this compartment.
using Debug = ConditionalDebug<true, "Hello world compartment">;
using namespace std::string_literals;

extern "C" ErrorRecoveryBehaviour
compartment_error_handler(ErrorState *frame, size_t mcause, size_t mtval)
{
	Debug::log("--- Opps ---");	
	//Debug::log("mcause: {}, pcc: {}", mcause, frame->pcc);
	auto [reg, cause] = CHERI::extract_cheri_mtval(mtval);
	//Debug::log("Error {} in register {}", reg, cause);
	return ErrorRecoveryBehaviour::ForceUnwind;
}

/// Thread entry point.
void __cheri_compartment("hello") say_hello()
{
	Debug::log("Hello world");
	const std::string S1 = "I am an walrus"s;
	Debug::log("Hello again");
	const std::string S2 = "I am the walrus"s;
	Debug::log("Hello once more");
}
``` 

gives
```
Hello world compartment: Hello world
Hello world compartment: Hello again
Hello world compartment: --- Opps ---
SUCCESS
```

but uncommenting the line to print mcause gives
```
Hello world compartment: Hello world
Hello world compartment: Hello again
Hello world compartment: SUCCESS
```

davidchisnall, 2024-06-19:
14 looks suspiciously like the length of a small string in libc++, so this looks like you’re hitting the case where the string uses a heap-allocated buffer. Is it possible that it’s requiring a global constructor? If so, you will need to run them using https://github.com/microsoft/cheriot-rtos/blob/main/sdk/include/c%2B%2B-config/global_constructors.hh

davidchisnall, 2024-06-19:
Oh, if the error handler is not running you may also be running out of stack space. The hello world example has a very small stack.

davidchisnall, 2024-06-26:
The C++ standard library APIs are not great for embedded environments.  They assume allocation failures will throw an exception, but in the absence of space for an unwinder that's a problem.  Allocation fails here and returns null, and then the string blindly copies the data into the (now invalid) buffer.  There's no way (in the standard API) for it to report failure except throwing an exception.

closed 2024-06-26 by davidchisnall