CHERIoT-Platform/cheriot-rtos #282: Allow per-shared objects [design proposal]

davidchisnall, 2024-08-12:
There are some use cases where it's convenient to pre-share a global.  We have a handful of places where it's useful to allow sharing objects where the (initial) sharing relationship is defined statically.  For example:

 - The restartable network stack has an epoch counter that various compartments use to determine the [re]initialisation status.
 - We expose various statistic counters in different compartments.
 - The config broker exposes a structure for each config item that contains an epoch counter and a pointer to the data.

Currently, these all share the same pattern: One compartment owns the global (or heap object, in a few cases) and hands out restricted capabilities to it.  This involves adding some extra function entry points that can hand out the capabilities and then dynamically setting the permissions.

In the loader, we have a generic mechanism for doing this currently, but it's restricted to memory-mapped I/O regions.  We explicitly check that these are not pointing to memory (with an explicit exemption for the allocator to let it access the hazards array).  I propose that we relax this and allow a set of shared objects to be defined in a shared-objects region of the program.

A shared object is not necessarily owned by a single compartment and so the proposed model separates out defining them and importing them.  Importing will use a slight modification of the existing MMIO macros (to eventually be replaced with compiler annotations).  

I propose that shared objects are created in xmake in a similar manner to threads (though collected from dependent builds, so we can define them in the build for a compartment).  Objects with the same name can be declared more than once, but must be declared with the same size.

The linker report should be extended to inspect the shared-object region and report the specific objects and sizes that are imported, rather than just addresses.

Pre-shared objects have the following restrictions:

 - As with MMIO objects, pointers to them cannot be used to initialise other globals.
 - They are always zero-initialised.
 - They must have globally-unique names within a firmware image.  The name does not have to be a valid C identifier.

rmn30, 2024-08-12:
Sounds good to me. I guess pre-shared objects can be writeable with the option to import them as RW, RO or WO? Asking because I think this is a use case for read-only global pointers to writeable objects that came up in the recent discussions about ISA permissions.

davidchisnall, 2024-08-12:
For MMIO, we have four bits on the import that embed load, store, load-mutable, and load-store capability permissions, so these will remain the same.

PhilDay-CT, 2024-08-12:
I might be misunderstanding the intent, but in the case of the config_broker the config_item+version isn't intended to be general shared data as such - a client should only be able to get access to a read-only copy if it can provide the appropriate sealed capability.

So with this change would auditing which compartments have the  READ_CONFIG sealed capability change to auditing which compartments have a RO access to each global config_item, with the config_broker being the only compartment with Write access to the global ?

davidchisnall, 2024-08-12:
> I might be misunderstanding the intent, but in the case of the config_broker the config_item+version isn't intended to be general shared data as such - a client should only be able to get access to a read-only copy if it can provide the appropriate sealed capability.

I was not 100% sure about this use case.

> So with this change would auditing which compartments have the READ_CONFIG sealed capability change to auditing which compartments have a RO access to each global config_item, with the config_broker being the only compartment with Write access to the global ?

Yes, exactly.  You'd need to build the broker compartment with a list of all of the pre-shared objects that it references, which might be more complex than the current code.

PhilDay-CT, 2024-08-12:
Might not be the best example for this - the broker was designed so that it doesn't have to have any knowledge of what config items exists, so it can be simply added into a system as-is.   Adding a new config item just requires the appropriate set of sealed capabilities to be defined in a publisher, parser, and subscriber(s).

commit 7414f2fd (2024-08-13): Add initial support for pre-shared globals.

This provides a mechanism to statically define globals that can be
shared between compartments statically.  As with MMIO imports, these can
be imported with restricted permissions.

They currently show up in the audit log as MMIO imports.  This is not
ideal, we should plumb their names through into the audit log so that
cheriot-audit can check that all references to a pre-shared object refer
to exactly one object.

Fixes #282
commit e6187e93 (2024-08-14): Add initial support for pre-shared globals.

This provides a mechanism to statically define globals that can be
shared between compartments statically.  As with MMIO imports, these can
be imported with restricted permissions.

They currently show up in the audit log as MMIO imports.  This is not
ideal, we should plumb their names through into the audit log so that
cheriot-audit can check that all references to a pre-shared object refer
to exactly one object.

Fixes #282
commit 37e5db09 (2024-08-14): Add initial support for pre-shared globals.

This provides a mechanism to statically define globals that can be
shared between compartments statically.  As with MMIO imports, these can
be imported with restricted permissions.

They currently show up in the audit log as MMIO imports.  This is not
ideal, we should plumb their names through into the audit log so that
cheriot-audit can check that all references to a pre-shared object refer
to exactly one object.

Fixes #282
PR #283 (2024-08-14): Add initial support for pre-shared globals.
This provides a mechanism to statically define globals that can be shared between compartments statically.  As with MMIO imports, these can be imported with restricted permissions.

They currently show up in the audit log as MMIO imports.  This is not ideal, we should plumb their names through into the audit log so that cheriot-audit can check that all references to a pre-shared object refer to exactly one object.

Fixes #282
closed 2024-08-14 by davidchisnall