CHERIoT-Platform/cheriot-rtos #307: Need a way to leave space at the front of a firmware image

sleffler, 2024-10-04:
I need to prepend firmware images with a data structure for use by OpenTitan's secure boot mechanism. To this end I set board.instruction_memory.start to be offset (by 896 bytes). But it turns out the default behaviour of lld is to try to write a copy of the ELF PHDRS at the front of the image if there is sufficient space "below" the first load segment: see [here](https://github.com/CHERIoT-Platform/llvm-project/blob/cheriot/lld/ELF/Writer.cpp#L2515) and [here](https://github.com/CHERIoT-Platform/llvm-project/blob/73a46080bd75229c31398414370a8ff2a5d51ec8/lld/ELF/LinkerScript.cpp#L1249). Given the 896 byte offset, space is available and the PHDRS are written, resulting in an image that is unusable. My solution was to add the -n option to the link cmd line:
```
index bac906e..da170b7 100644
--- a/sdk/xmake.lua
+++ b/sdk/xmake.lua
@@ -747,7 +747,7 @@ rule("firmware")
                                table.insert(objects, dep:targetfile())
                        end
                end)
-               batchcmds:vrunv(target:tool("ld"), table.join({"--script=" .. linkerscript, "--relax", "-o", target:targetfile(), "--compartment-report=" .. target:targetfile() .. ".json" }, objects), opt)
+               batchcmds:vrunv(target:tool("ld"), table.join({"-n", "--script=" .. linkerscript, "--relax", "-o", target:targetfile(), "--compartment-report=" .. target:targetfile() .. ".json" }, objects), opt)
                batchcmds:show_progress(opt.progress, "Creating firmware report " .. target:targetfile() .. ".json")
                batchcmds:show_progress(opt.progress, "Creating firmware dump " .. target:targetfile() .. ".dump")
                batchcmds:vexecv(target:tool("objdump"), {"-glxsdrS", "--demangle", target:targetfile()}, table.join(opt, {stdout = target:targetfile() .. ".dump"}))
```
This disables the writing of PHDRS; the other effects of -n seem irrelevant: no page alignment of sections  (done already by the linker script), and link against static libraries.

Does anyone have a better suggestion?

davidchisnall, 2024-10-04:
I see no downsides to adding -n.

sleffler, 2024-10-04:
https://github.com/CHERIoT-Platform/cheriot-rtos/pull/308