CHERIoT-Platform/cheriot-rtos #322: Libraries should be compiled with CHERIOT_NO_AMBIENT_MALLOC

PhilDay-CT, 2024-10-25:
Currently it's possible to define a malloc capability in code that gets compiled into a library, in effect giving a library a heap quota even if the compartment that includes the library is defined not to have heap quota.

xmake rules for libraries should be updated to compile with CHERIOT_NO_AMBIENT_MALLOC defined

https://github.com/CHERIoT-Platform/cheriot-rtos/blob/b9f206a7c9b138e762b51f539d085a4d7377a516/sdk/xmake.lua#L155

davidchisnall, 2024-10-25:
Putting the define in will still make it possible (though you can prevent it via policy and we should add some rules for doing this), but it will make the default sensible.

davidchisnall, 2024-11-07:
When we're building a compartment, I think the compiler puts the compartment name in a `#define`, so an alternative here would be for `stdlib.h` to flip the default depending on whether the define is set.