CHERIoT-Platform/cheriot-rtos #368: Using == on std::string causes undefined symbol: __library_export_libcalls_memcmp

stormeuh, 2024-12-05:
When I use `==` or `compare` between objects of type `std::string` the linker complains as follows:
```
error: ld.lld: error: undefined symbol: __library_export_libcalls_memcmp
>>> referenced by hello.cc
>>>               build/cheriot/cheriot/debug/hello.compartment:()
```

I get this error with the following minimal example.
hello.cc:
```
#include <compartment.h>
#include <debug.hh>
#include <string>

using Debug = ConditionalDebug<true, "Hello world compartment">;

/// Thread entry point.
void __cheri_compartment("hello") say_hello()
{
	std::string first = "first";
	std::string second = "second";
	if (first == second) {
		Debug::log("Equal!");
	}
}
```
xmake.lua:
```
set_project("CHERIoT Hello World")
sdkdir = "../../sdk"
includes(sdkdir)
set_toolchains("cheriot-clang")

option("board")
    set_default("sail")

compartment("hello")
    add_deps("freestanding", "debug", "string")
    add_files("hello.cc")

firmware("hello_world")
    add_deps("hello")
    on_load(function(target)
        target:values_set("board", "$(board)")
        target:values_set("threads", {
            {
                compartment = "hello",
                priority = 1,
                entry_point = "say_hello",
                stack_size = 0x300,
                trusted_stack_frames = 3
            }
        }, {expand = false})
    end)
```

davidchisnall, 2024-12-05:
Memcmp is exported with the mangled name. We should probably just stop doing that and export them h mangled, as we do with the atomics.

resistor, 2024-12-06:
Is this what you had in mind? https://github.com/resistor/llvm-project-1/tree/memcmp

davidchisnall, 2024-12-06:
I think so, there’s also some stuff in the back end that explicitly handles these (memcpy is expanded multiple times in LLVM, we need to turn it into a libcall if it’s expanded late and that also includes mangling).

It looks as if we’ve missed one case for memcmp, and I suspect it’s easier to just export our freestanding lib calls with unmangled names than to keep fixing things in the compiler.

As a transition thing, I think we can modify the freestanding lib to export the symbols under both names, then eventually remove the mangled one.

stormeuh, 2024-12-06:
Thank you all for the quick triage!

The patched compiler solves the issue, though I'd like to know if @davidchisnall's freestanding lib patch would be easier in my use case.

I'm using CHERIoT to create a project for students, which they can then work on in the devcontainer.
Replacing the compiler in the devcontainer is fine on x86, but I'd like an image for ARM as well and I'm not fond of compiling clang through qemu...

Do you have any advice?

rmn30, 2024-12-06:
The devcontainer has an image for ARM64 and once a fix is merged we'll rebuild it. Will that work?

Exciting to hear you're using CHERIoT with students. Don't hesitate to get touch if you need any more help!

davidchisnall, 2024-12-06:
I think we'll do the freestanding lib patch soon and then slowly migrate the compiler.  Please can you test #370, it should allow the old or new compiler to work.  We can gradually then migrate to not mangling these symbols.

A few other folks are interested in projects for student, it would be great if you could join [the Signal chat](https://signal.group/#CjQKIElxAs3t3MUEMOEmQEuMHRK4rErUk2xVeFzjAjFXAShzEhCK9qQwEMFKGLGZnCjrQ7zm) and share what you're working on!

davidchisnall, 2024-12-09:
Ah, yes, that will cause problems.  I'll add the other bit of the patch to make it explicitly set symbol names and then it should be agnostic to the compiler.  Thanks!

davidchisnall, 2024-12-09:
Please can you test the updated version of #370?  This should now work with both versions of the compiler.

commit 3de03513 (2024-12-06): Export freestanding symbols as unmangled.

Mangling these is painful for the compiler and we've missed at least one
place where it needs to happen.  We don't mangle the atomics, so it's a
good idea to also not mangle the freestanding ones.

As a transitional measure, export the functions as both their mangled
and unmangled forms.

The next step is to make the mangled form the compatibility definition
and use `asm` in the header to export these as non-mangled by default
for direct calls, then finally to remove `compat.S`.

Once this is merged, the compiler is free to start emitting `mem*` as
li [... cut]
commit 3ca765e0 (2024-12-09): Export freestanding symbols as unmangled.

Mangling these is painful for the compiler and we've missed at least one
place where it needs to happen.  We don't mangle the atomics, so it's a
good idea to also not mangle the freestanding ones.

As a transitional measure, export the functions as both their mangled
and unmangled forms.

The next step is to make the mangled form the compatibility definition
and use `asm` in the header to export these as non-mangled by default
for direct calls, then finally to remove `compat.S`.

Once this is merged, the compiler is free to start emitting `mem*` as
li [... cut]
commit 118e22c3 (2024-12-09): Export freestanding symbols as unmangled.

Mangling these is painful for the compiler and we've missed at least one
place where it needs to happen.  We don't mangle the atomics, so it's a
good idea to also not mangle the freestanding ones.

As a transitional measure, export the functions as both their mangled
and unmangled forms.

The next step is to make the mangled form the compatibility definition
and use `asm` in the header to export these as non-mangled by default
for direct calls, then finally to remove `compat.S`.

Once this is merged, the compiler is free to start emitting `mem*` as
li [... cut]
commit b23d7bc7 (2024-12-09): Export freestanding symbols as unmangled.

Mangling these is painful for the compiler and we've missed at least one
place where it needs to happen.  We don't mangle the atomics, so it's a
good idea to also not mangle the freestanding ones.

As a transitional measure, export the functions as both their mangled
and unmangled forms.

The next step is to make the mangled form the compatibility definition
and use `asm` in the header to export these as non-mangled by default
for direct calls, then finally to remove `compat.S`.

Once this is merged, the compiler is free to start emitting `mem*` as
li [... cut]
closed 2024-12-09 by davidchisnall