CHERIoT-Platform/cheriot-rtos #381: fail-simulator-on-error can exit with success

nwf, 2024-12-18:
If the [`fail-simulator-on-error.h` error handler](https://github.com/CHERIoT-Platform/cheriot-rtos/blob/main/sdk/include/fail-simulator-on-error.h) itself crashes -- by, say, exhausting the stack during a `ConditionalDebug::log` call -- then the simulation can end with a successful result, because we don't reach the `simulation_exit(1)` call.

I'm not quite sure what to do about this, but it sure is sad.  It'd have been nice if HTIF and friends had allowed separately setting the exit status and commanding the exit, but alas.

rmn30, 2024-12-18:
Yes, I've known this for a while. The Hello World example was exiting that way due to its small stack and returning from the entry point causing a trap on returning to null. Thankfully you fixed that by setting up the the thread entry point so it returns cleanly into the switcher and then thread exit.

One potential solution would be to add a stackless error handler and use `CHERIOT_DURING` etc. in the `compartment_error_handler`! This would require linking with unwind library.

nwf, 2025-01-22:
Belatedly: the situation could be improved (though not completely fixed), if `simulation_exit()` could be a library call, rather than a cross-compartment call.