CHERIoT-Platform/cheriot-rtos #480: Does not link anymore on the Arty

hlef, 2025-03-16:
Since 3cffc5c the Arty does not link anymore:

```
error: ld.lld: error: undefined symbol: __export_mem_kunyan_ethernet
>>> referenced by __cpp_firewall.cc.cc
>>>               build/cheriot/cheriot/release/Firewall.compartment:()
>>> referenced by __cpp_firewall.cc.cc
>>>               build/cheriot/cheriot/release/Firewall.compartment:()

ld.lld: error: undefined symbol: __export_mem_kunyan_ethernet_end
>>> referenced by __cpp_firewall.cc.cc
>>>               build/cheriot/cheriot/release/Firewall.compartment:()
```

Reverting to the old JSON does the trick. I could not pinpoint the issue, unfortunately, so applying the old JSON on my system as a workaround for now.

rmn30, 2025-03-19:
I think this is a mistake in`sdk/boards/ibex-arty-a7-100.patch`. Probably `path: "/kunyan_ethernet"` should be `path: "/devices/kunyan_ethernet"`?

davidchisnall, 2025-03-19:
Yup, that looks like an error.

PR #486 (2025-03-19): Fix incorrect path for kunyan_ethernet device in arty-a7 board patch.
Fixes #480 .
commit ca2d1789 (2025-03-21): Fix incorrect path for kunyan_ethernet device in arty-a7 board patch.

Fixes #480.
closed 2025-03-19 by rmn30