CHERIoT-Platform/cheriot-rtos #499: RISC-V Privilege Spec says HPM counters are uninitialized at boot

nwf, 2025-03-28:
https://five-embeddev.com/riscv-priv-isa-manual/Priv-v1.12/machine.html#hardware-performance-monitor in particular sayeth, referring to m{cycle,instret,hpmcounter*},

> The counter registers have an arbitrary value after the hart is reset, and can be written with a given value.

We should probably zero these early in boot?

davidchisnall, 2025-03-31:
I think we can document a requirement that boot ROM code should zero them if the hardware does the stupid thing.

davidchisnall, 2025-04-05:
(I suspect the reason that the spec says this is that it adds a state machine to handle the reset state and it’s easier to just shove some boot ROM in some cores than a load of state machines)