CHERIoT-Platform/cheriot-rtos #559: Refactor switcher error path

nwf, 2025-07-25:
@stellamplau and @vmurali point out that [the switcher's exception handler return path](https://github.com/CHERIoT-Platform/cheriot-rtos/blob/main/sdk/core/switcher/entry.S#L1591-L1727) is a lot of fiddly code that doesn't need to run with ASR.  Moreover, it complicates formal verification: the state of the GPRs after a trip through the switcher's exception path is not necessarily related to the contents of the trap frame contents as of handler return, because the switcher's loads are potentially data racing other threads.

I think we can...
1. `mret` into the fault handler instead of [the `cjalr` we do now](https://github.com/CHERIoT-Platform/cheriot-rtos/blob/cdfb5b8977fbb6db12ddc94a70ee313a7b22787f/sdk/core/switcher/entry.S#L1590), at the cost of briefly requiring IRQs to be deferred (if they aren't already).

2. Have the switcher expose an IRQ-disabled ASR-bearing context-restore function (which needs to perform a load and use `mepcc` to do its thing, AFAICT, because RISC-V can implement `longjmp` in userspace, but can't implement `setcontext()` without CSRs).

3. Wrap the compartment error handler invocation in a library function `void wrap_error_handler(ErrorFrame *, size_t mcause, size_t mtval, enum ErrorRecoveryBehaviour (*compartment_error_handler)(ErrorFrame *, size_t, size_t))` that takes a spilled frame (from the switcher, still), calls the (unsealed!) provided error handler, and, if that requests `InstallContext`, does the dance of reconstituting the frame's PCC using the provided error handler function pointer's bounds and then uses the above exposed primitive.

rmn30, 2025-07-25:
Do we allow interrupt-disabled error handlers? Just thinking that calling it via an unsealed pointer means you'd have to manage that manually.

nwf, 2025-07-25:
> Do we allow interrupt-disabled error handlers? Just thinking that calling it via an unsealed pointer means you'd have to manage that manually.

No, all error handlers run with IRQs enabled.

vmurali, 2025-07-28:
I think the alternative design suggestion is wrong (@stellamplau?). The property I need is that the supplied ucontext object is in the stack so that it doesn't get clobbered by another thread.

stellamplau, 2025-08-01:
Just leaving a note here that the switcher can also write to globals on the [exception handler entry path](https://github.com/CHERIoT-Platform/cheriot-rtos/blob/94f37aeb68e0ca102283389ae07ef4ac37196aa8/sdk/core/switcher/entry.S#L926). This means that user code currently needs to be written/audited defensively to assume that anytime an exception can occur in any thread where `csp` points to its globals, a register spill frame worth of memory can be clobbered (with non-thread-local effects).