CHERIoT-Platform/cheriot-rtos #560: Side effect in switcher's compartment-call exception path

stellamplau, 2025-07-25:
@vmurali and I noticed that an exception in the [switcher's compartment call path](https://github.com/CHERIoT-Platform/cheriot-rtos/blob/cdfb5b8977fbb6db12ddc94a70ee313a7b22787f/sdk/core/switcher/entry.S#L325) can lead to the side effect of storing registers to non-stack memory if the caller's `csp` is non-local. In this case, the switcher will first spill registers to non-local memory, then throw an exception and unwind the stack. The writes to non-local memory along this path seem like unintended side effects. 

(For context: this is motivated similarly to [#559](https://github.com/CHERIoT-Platform/cheriot-rtos/issues/559), where reasoning about the switcher would be simplified if it only ever accessed registers, the stack, and read-only memory).

A concrete proposal would be to check the integrity of csp first, before attempting to spill registers (in other words, swap the order of [.Lswitch_entry_first_spill](https://github.com/CHERIoT-Platform/cheriot-rtos/blob/cdfb5b8977fbb6db12ddc94a70ee313a7b22787f/sdk/core/switcher/entry.S#L325C1-L325C27) with [.Lswitch_csp_check](https://github.com/CHERIoT-Platform/cheriot-rtos/blob/cdfb5b8977fbb6db12ddc94a70ee313a7b22787f/sdk/core/switcher/entry.S#L356)).

davidchisnall, 2025-07-26:
I don’t think this is a problem. Anything that happens in this context is logically in the context of the caller.

This is a code-size optimisation to avoid every caller needing to spill these registers. The switcher can’t overwrite store-local, these spills occur exactly as if the caller had done them.

Adding additional checks here would slow down the non-malicious path. The current design is optimised for the non-malicious case and makes the malicious case. If the caller does something that makes this trap, we assume it’s malicious and force unwind out (you attack the switcher, you don’t get to retry immediately).

nwf, 2025-07-28:
> The fix is actually simple and not expensive. Just check if CSP points to a stack (CSP canStore Local i.e. has StoreLocal permission) before trying to check if stack has enough size for the callee.

Depending on what you mean... we already do that? `.Lswitch_csp_check` _is_ before `.Lswitch_stack_check_length` (which has to be after `.Lswitch_unseal_entry`, because we need to know the callee's minimum stack length, and so is actually fairly late in the cross-call path).  

But I think that isn't what you mean by "check if the stack has enough size for the callee".  I'd have said "enough size for the _call_", since the initial spills are defensive and on behalf of the _caller_.  In any case, I think your point is that we should move the `.Lswitch_cap_check` block to _before_ `.Lswitch_entry_first_spill`.  This is, in fact, quite easy: `.Lswitch_csp_check` operates entirely within dead registers (specifically, `tp` and `t2`) [^0] and nothing about its error path, `.Lcommon_force_unwind` depends on the spill area we created in `.Lswitch_entry_first_spill`.

[^0]: ... and in fact can trivially be shaved down to just one... with CHERIoT's architectural permission layout, where the numeric value of `LD|SD|MC|LM|SL|LG` fits in a 12 bit immediate, but not with the proposed spec's, where R (LD) is way over at `1 << 18`.  Bah humbug.  Perhaps we could exploit that LM requires R (LD).  Pardon the tangent.