CHERIoT-Platform/cheriot-rtos #563: [CHERI CSA Report] Tag-stripping store in sdk/core/loader/boot.cc

resistor, 2025-07-26:
The new CHERI clang-static-analyzer checkers are producing the following warning. This could very well be a false positive.

```
---
MainSourceFile:  '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
Diagnostics:
  - DiagnosticName:  clang-analyzer-cheri.CapabilityCopy
    DiagnosticMessage:
      Message:         Tag-stripping store of a capability
      FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
      FileOffset:      51603
      Replacements:    []
      Ranges:
        - FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
          FileOffset:      51595
          Length:          17
    Notes:
      - Message:         'void * __capability argument may be a pointer to capability'
        FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
        FileOffset:      51499
        Replacements:    []
        Ranges:
          - FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
            FileOffset:      51499
            Length:          3
      - Message:         'const void * __capability argument may be a pointer to capability'
        FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
        FileOffset:      51551
        Replacements:    []
        Ranges:
          - FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
            FileOffset:      51551
            Length:          3
      - Message:         'Assuming ''i'' is < ''n'''
        FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
        FileOffset:      51578
        Replacements:    []
        Ranges:
          - FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
            FileOffset:      51578
            Length:          5
      - Message:         Loop condition is true.  Entering loop body
        FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
        FileOffset:      51559
        Replacements:    []
      - Message:         Tag-stripping store of a capability
        FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
        FileOffset:      51603
        Replacements:    []
        Ranges:
          - FilePath:        '/workspaces/cheriot-rtos/sdk/core/loader/boot.cc'
            FileOffset:      51595
            Length:          17
    Level:           Warning
    BuildDirectory:  '/workspaces/cheriot-rtos'
...
```

vmurali, 2025-07-28:
Points to https://github.com/CHERIoT-Platform/cheriot-rtos/blob/main/sdk/core/loader/boot.cc#L1469. Why does this code exist?

resistor, 2025-07-29:
https://github.com/CHERIoT-Platform/cheriot-rtos/pull/568

nwf, 2025-07-30:
For my edification, would building `loader/boot.cc` with the "baremetal" ABI mean that the compiler wouldn't try to emit libcalls to `memcpy`?

davidchisnall, 2025-07-31:
Should have been closed by merging #568, I think?