CHERIoT-Platform/cheriot-rtos #564: [CHERI CSA Report] Bad pointer alignment cast in sdk/lib/queue/queue_compartment.cc

resistor, 2025-07-26:
The new CHERI clang-static-analyzer checkers are producing the following warning. This could very well be a false positive.

```
---
MainSourceFile:  '/workspaces/cheriot-rtos/sdk/lib/queue/queue_compartment.cc'
Diagnostics:
  - DiagnosticName:  clang-analyzer-optin.portability.PointerAlignment
    DiagnosticMessage:
      Message:         'Pointer value aligned to a 4 byte boundary cast to type ''RestrictedEndpoint * __sealed_capability'' with 8-byte capability alignment'
      FilePath:        '/workspaces/cheriot-rtos/sdk/lib/queue/queue_compartment.cc'
      FileOffset:      1321
      Replacements:    []
    Notes:
      - Message:         'Calling ''unseal'''
        FilePath:        '/workspaces/cheriot-rtos/sdk/lib/queue/queue_compartment.cc'
        FileOffset:      5119
        Replacements:    []
        Ranges:
          - FilePath:        '/workspaces/cheriot-rtos/sdk/lib/queue/queue_compartment.cc'
            FileOffset:      5119
            Length:          26
      - Message:         'Pointer value aligned to a 4 byte boundary cast to type ''RestrictedEndpoint * __sealed_capability'' with 8-byte capability alignment'
        FilePath:        '/workspaces/cheriot-rtos/sdk/lib/queue/queue_compartment.cc'
        FileOffset:      1321
        Replacements:    []
        Ranges:
          - FilePath:        '/workspaces/cheriot-rtos/sdk/lib/queue/queue_compartment.cc'
            FileOffset:      1321
            Length:          60
    Level:           Warning
    BuildDirectory:  '/workspaces/cheriot-rtos'
...
```

davidchisnall, 2025-07-26:
This looks like maybe the analysers don’t know what a sealed capability is?

resistor, 2025-07-26:
> This looks like maybe the analysers don’t know what a sealed capability is?

Seems plausible.

resistor, 2025-07-29:
Resolved by https://github.com/CHERIoT-Platform/llvm-project/pull/204

closed 2025-07-29 by resistor